iceman1001/ChameleonMini-rebootedGUI #1: Link to some old GUI source code

iceman1001, 2017-12-19:
https://pastebin.com/3Y6SGFi9

In order to keep track of how they called the mfkey code...

iceman1001, 2018-01-08:
There are some nasty bugs in this part.  
1. the define,  not sure if the XOR applies on "key" or on the result of the define.
2. size != len,   if len == 208, and tobeencfilename doesn't always downloads 208,  but say 150, then the loop will overwrite stuff. Same goes for the **iso14443acrc** function.
```
#define genFun(size, key, i) size + key + i - size / key
void ComPass(uint8_t *toBeEncFileName, int key, int len)
{
    uint8_t newFileName[275] = { 0 };
    memcpy(newFileName, toBeEncFileName, len);
    int i, s, t, size = len;
 
    for (i = 0; i < size; i++)
    {
        s = newFileName[i];
        t = genFun(size, key, i) ^ s;  // 加密  
        toBeEncFileName[i] = t;
    }
}
```

iceman1001, 2018-01-08:
All those bugs clear,  and I got the sniff keys to work :)

bogiton, 2018-01-08:
Very good job man! :)

iceman1001, 2018-01-09:
Its all pushed to the repo! You can start testing it and take it to the next level :)

With this,  the old GUI is more or less matched with this project.  I will close this issue.

closed 2018-01-09 by iceman1001