iceman1001/ChameleonMini-rebootedGUI #6: new release v1.0

iceman1001, 2018-01-07:
And here I open up for the first release of the Rebooted GUI.

It should working with **both** old rebooted firmware  **Chameleon-new-1.0**  and the updated firmware from here [](https://github.com/iceman1001/ChameleonMini-rebooted)

iceman1001, 2018-01-09:
We see a time when old and new GUI will work together, but in the long run, the old one will disappear since the firmware here has fixes and enhancments and those will break functionality with old GUI.

So,
old gui + old fw  = ok
old gui + new fw = ok
new gui + old fw = ok
new gui + new fw = ok

---We will try to have backwardsfunctionality with old firmware.  Its easy to detect what it can do, and we can compensate for it in the GUI..
meaning the following should apply for a long time.
new gui + old fw = ok
new gui + new fw = ok

iceman1001, 2018-01-21:
@bogiton  I did a click-once deployment.   So, if you are fine with current version of the GUI,  we might aswell create this release :)

bogiton, 2018-01-21:
Cool stuff this click-once deployment!
I was thinking that maybe we should wait a couple of days to see if any bug/problem come out and then go forward with the release! :)

iceman1001, 2018-01-21:
yeah, I did some minor fixes in the looping. 
Changing the mfkey at the moment to use parallell instead,  got stuck with progressbar reporting..

The default timeout should also be a settings / appconfig value.

bogiton, 2018-01-21:
Parallel as in async? Yeah, I just remembered the unknown Calculation_times() function in the sample gui code you provided, which was used for the progress division.
Or we could just say something like for the timeout: "If not set, default interval is 2sec".

iceman1001, 2018-01-22:
@bogiton  you can proof-read the draft?  If ok,  we publish it.

bogiton, 2018-01-22:
Go for it :)

iceman1001, 2018-01-22:
v1.0 is released :)
https://github.com/iceman1001/ChameleonMini-rebootedGUI/releases

closed 2018-01-22 by iceman1001