iceman1001/ChameleonMini-rebootedGUI #7: GUI rewamped

iceman1001, 2018-01-09:
I couldn't stand it,  so I changed the layout and design a bit. 

The GUI is more compacter and less buttons now, with easier reads of findings like mfkeys...

I can imagine in future that you select a tagslot panel,  and have only five buttons instead of forty.

[image]
[image]

bogiton, 2018-01-10:
Apart from the 40 buttons, I like it! :)
We could only leave the "Apply" button for each slot plus one named "More..." that will pop-up a dialog with the additional actions (Upload, Download, Mfkey, Clear).

iceman1001, 2018-01-10:
I liked your mockup,   when I look at the gui I think that we could "select"  a tagslot,  and with that we only need five buttons.  If we then allow to have "several" tagslots selected,  we could let a button iterate all selected slots and execute its function.   That would enable us to get ride of many buttons,  and still be able to apply one button press on several slots.    Like  "clear" today, needs 8 seperate clicks.
My idea,   select and one click.

Not to mention remove  "dump" from  "upload dump" and "download dump".

But all of this is a starting point.   No need to feel we have to have this design.   Feel creative!  and make is awesume!

bogiton, 2018-01-16:
What do you think of something like this?

[image]

iceman1001, 2018-01-16:
Yeah baby!   Lovin' it!
Maybe checkboxes,  so we can select several at once :)

iceman1001, 2018-01-16:
We also need to figure out a dynamic way of having mem_sizes on device.  We can't reserve up 4kb per tag as we do today.   Like make 2 - 4 slots can be 4Kb max.   8*4=32kb..    4*4 + 4*1 = 20kb..
The UL / UlC / ULEV1 / NTAG is much smaller..

bogiton, 2018-01-16:
Hmm checkboxes would be useful for applying/clearing multiple slots, but there has to be only one slot selected for the other three buttons (mfkey, upload, download). Also, I went for the radio buttons to actually set an active slot (settingmy=X) upon selection/clicking.

Dynamic allocation sounds like a very good way to save memory space. Maybe we have to try it.

iceman1001, 2018-01-16:
I think we can have multiple checkboxes for  mfkey,   its just a loop around each box ;)
upload seems stupid to have multiple..
download could also work with multiple..

bogiton, 2018-01-16:
Is it possible to have set MF_DETECTION to multiple slots? Then yeah, multiple mfkey makes sense. On the other hand, the multiple-download should require a pre-defined directory to save the dumps (launching up to 8 save dialogs is ugly :P)
I believe it would be useful to also provide a way to switch to the active slot through the GUI. I'm thinking of how this could be implemented with the checkboxes. Maybe if only one checkbox is checked, then set this slot as active?! Kinda confusing.

iceman1001, 2018-01-16:
I think we should have  pageview (like we talked about)  and have a settings tag.  Where we set default folder for upload / download.    
The MF_DETECTION is possible for multiple tagslots.  You can pretend to be other UIDs...

Why do we "need" to set a slot active?  Whats the real point?

bogiton, 2018-01-16:
I was also thinking about a settings dialog, probably accessible through a top menu. And then maybe add a few more dialogs there, like custom command sending, version/rssi info, etc.

If the button action is not set to CYCLE_SETTINGS/SWITCHCARD, the user cannot select a slot by pressing the button. The only way would be to connect through a terminal and issue the settingmy command. That's why I thought it should be provided.

iceman1001, 2018-01-18:
I like the tab-interface,   instead of a menubar with seperate forms.

We should either contact this dude or get inspiration from his GUI..
http://35b0.de/wp-content/uploads/2017/12/chameleonmini-gui.jpg

bogiton, 2018-01-18:
Hmmm yeah, tab-interface is better than the menustrip.
Alright, I will try these days to implement most of what we discussed and add it to my fork. Then we can test it out and eventually push it to yours.

iceman1001, 2018-01-18:
Go for it!

bogiton, 2018-01-21:
https://github.com/bogiton/ChameleonMini-rebootedGUI

I think it contains what we discussed, plus some more goodies.
Let me know what you guys think of it. If you like it, I will do a PR.
Tried to test it out as much as I could, but there are still bugs left in it possibly.
One thing in the TODO list is to use the progressbar more :P

iceman1001, 2018-01-21:
Great,
its a good stepping stone to more.

Like smaller checkbox for (select all, none) instead of buttons.
and minor textual things like  "Tag Slot" -"Slot", 

It has a much nice feeling!

bogiton, 2018-01-21:
I'm glad you like it.
Do you want me to PR now and work on the rest of the changes here?

iceman1001, 2018-01-21:
Go ahead with the PR!

iceman1001, 2018-01-22:
Now that we got tabs (yay!)
What do you @bogiton  think about having a tab for dump-manangement?   Like bin-diff?  converting?

bogiton, 2018-01-22:
That would be epic!
I'm gonna check for a nice hex editor to include.

iceman1001, 2018-01-22:
yeah, that would be excellent.   Which bin-diff / hexeditors is open-sourced??

bogiton, 2018-01-22:
The Be.HexEditor (https://sourceforge.net/projects/hexbox/files/hexbox/) seems to be the most popular C# open source hex editor (control).
I suppose we could use two of those controls to get our bin-diff functionality.

iceman1001, 2018-01-22:
yeah,  I noticed the same,  searching github...  MIT-license works with GPL v2.0.

iceman1001, 2018-01-22:
I suggest we close this issue,   and create a dedicated one for the bin-diff idea.

bogiton, 2018-01-22:
Alright, sounds good.

closed 2018-01-22 by iceman1001