iceman1001/ChameleonMini-rebootedGUI #8: download dump file too big 64kb

iceman1001, 2018-01-09:
I tested to upload a tagdump which worked nice,  then I tried to download it and compare it with my original dump file.
All well, but the dump file is 64kb large.  Not at all the 1kb which I was expecting.

Something in the xmodem received function is bad

iceman1001, 2018-01-09:
Aha,  well.. maybe its in the firmware..

The download of dumps is done via xmodem and  this function inside  memory.c  MemoryDownloadBlock( buffer, blockaddress, bytecount)

It uses the define MEMORY_SIZE_PER_SETTING as bound to how much to send...

bogiton, 2018-01-10:
Another solution would be to limit the number of bytes that the GUI receives/sends by first querying the current MEMSIZE of the selected slot.

iceman1001, 2018-01-10:
yup. go for that solution :)

bogiton, 2018-01-10:
Done. Not very optimal, but it works.

iceman1001, 2018-01-10:
what to do?  At least we have something that works on offical revg,  offical rebooted and iceman rebooted.  Who can complain about that?  First get it to work,  then make it nicer.

closed 2018-01-10 by iceman1001