iceman1001/ChameleonMini-rebootedGUI #9: COM port polling

bogiton, 2018-01-11:
The old GUI sends a keep-alive command (VERSIONMY) every 2 seconds to the device to check if it is connected or not. It also only opens the port to send a command (and receive the result) and then closes it.
In our version, once the device is connected, the port stays open as long as the program runs.

Do you think we should follow the old GUI's method of polling and opening-closing the COM port?

iceman1001, 2018-01-11:
well,  if we open /close,   then we can connect / reconnect quite easy.
With always open,  the comport will "break"  and we need to make a new connection...

No need to be connected all the time.

bogiton, 2018-01-11:
Alright then, we follow the old GUI's method. :)

iceman1001, 2018-01-11:
Or we poll ourselfs with a timer and/or a failed attempt,  then we try to reconnect again.
not sure which is the better option.   Have you looked at http://35b0.de/chameleonmini-gui/
His GUI is nice,  and we should also deploy the GUI as a click-once application,  where we can push out updates so ppl doesn't have to download the source code and compile..  Maybe an other issue/idea.

However,  I noticed that we do push directly to the repo.
It would be better if we both "fork" this repo and from our own forks make pull requests.
It is easier to maintain then.

iceman1001, 2018-01-22:
time to close?