iceman1001/ChameleonMini-rebootedGUI #11: SAK return (08) or (88) and unlock command.

kevin2008-01, 2018-01-15:
Please implement choice for SAK return (08) or (88).
Please implement choice for lock or unlock command for news detection by readers.

Thank you very much in advance.

iceman1001, 2018-01-15:
Lets see,
choice of SAK and ATQA  could be doable. I can see this would be nice to have.
lock / unlock command for new detect?  Sorry, I don't understand.

How knows if there will be more buttons?  And if someone think they want to add these ideas then that would be great.

kevin2008-01, 2018-01-15:
New reader detect magic command and demand SAK88 not 08, right see vigik challenge.

If you implement this choice, is magic.... see you soon.

kevin2008-01, 2018-01-22:
[image]

kevin2008-01, 2018-01-22:
this reader detect Chinese tag, with magic command or not. All chenese tag return SAK 08, if you write 88 in block0, but original tag return SAK 88.

kevin2008-01, 2018-01-22:
[image]

iceman1001, 2018-03-28:
1. unlock / lock  ->  Enable / disable magic command support?
2. custom SAK / ATQA.     would need three bytes in stored in flashmem.  plus a bool for enable / disable this function. 

Both of these functions would need firmware support first,  before GUI can support it.

iceman1001, 2019-02-21:
Looks like @kevin2008-01 got his wish granted by @bogiton   https://github.com/iceman1001/ChameleonMini-rebooted/pull/73

bogiton, 2019-02-21:
For the time being you can set the ATQA/SAK from the Serial tab. For example:
`SAKMY=88`
would set the SAK to 88 for the active tag.
Type `SAKMY?` to print the current value. (@iceman1001 The naming of that command is actually very funny, so maybe we have to remove the MY extension in future versions... :P )
As for the magic support, you can enable/disable it by setting the SUPPORT_MF_CLASSIC_MAGIC_MODE define before the re-compilation of the firmware. By default it is set to disabled.

iceman1001, 2019-02-21:
sak my 88... well luckily we are not teenagers with bad sense of humor.
Didn't we say we were going to drop those "MY" extensions once we got things under control.
With the firmware naming in place,  we can be more free.  Gui will recoqnise old style,  so that will need to be adapted.

iceman1001, 2019-03-12:
@kevin2008-01   Have you tested latest GUI & Firmware?

closed 2019-03-12 by iceman1001