iceman1001/ChameleonMini-rebootedGUI #13: v1.0.0 GUI - click once deployment

iceman1001, 2018-01-21:
There were some talk about binary releases or distro for those who can't compile the source for the GUI.
I did a click-once deployment, where it downloads and installed it from the link below.

It also checks for updates,  so yes, it calls home,  read on click-once what it means,  its the Microsoft built-in option.  If update is found, it updates the software and restarts it.   

Go ahead and test it out!

REF:
http://www.icesql.se/download/ChameleonMiniGUI/publish.htm

kevin2008-01, 2018-01-22:
unable to connect with chameleon.
(stock firmware 1.0)

iceman1001, 2018-01-22:
Do you have other usb-devices connected?   The com-port selection could be buggy

iceman1001, 2018-01-22:
pushed some changes,  close the gui and restart it,  that will trigger the update

kevin2008-01, 2018-01-22:
1 - install GUI, exit Gui, restart GUI.
2 - unable to connect to the chameleon.
(stock firmware new 1.0) (com8) (disable signed driver) (win 8.1 - 64)
3 - Driver is correctly installed and no problem.
4 - start original GUI and working.

iceman1001, 2018-01-22:
Uninstall?   It should be the normal way of un-installning a software in Windows,  nothing out-of-norm there.
Which device are you using,  revE rebooted or revG?  
And do you have more devices connected ?

kevin2008-01, 2018-01-22:
1 - i use setup.exe and accept many warning, and many authorization.
2 - when i launch GUI connect to your server, it's ok.
3 - When i click connect, i have this message (unable to connect to the chameleon).
4- I have revE rebooted with original firmware (original firmware: version: new chameleon 1.0).
5 - No more device attached, only rebooted on com8 or com3.
6 - Rebooted work fine with original sotware, and no problem with the driver.
7 - When i use the rebooted, i boot windows 8.1 with unsigned driver permission.

I'm not sure, but the GUI can't detect rebooted.

kevin2008-01, 2018-01-22:
1 - GUI can't detect, and connect error when i click connect. Disconnect is black.
2 - Changing port is my first action. (driver attribute com8, i change with com3), com5 is occupied by pm3 driver.

iceman1001, 2018-01-22:
This is strange.  I use rebooted device,  with original firmware and it finds the device. 

The application needs to download some dependencies to your computer like dot.net v4.5.2

kevin2008-01, 2018-01-22:
i see, but Framework 4.5.2 or higher is installed on you computer

kevin2008-01, 2018-01-22:
now i update the gui, and no working

kevin2008-01, 2018-01-22:
question: the driver is LUFA-CDC ... ?

iceman1001, 2018-01-22:
LUFA-CDC == normal mode ,  yes

kevin2008-01, 2018-01-22:
i have this

bogiton, 2018-01-22:
@kevin2008-01 Can you please check what name does the device has in the device manager? It should be listed under the "Ports (COM & LPT)".

kevin2008-01, 2018-01-22:
[image]
#
[image]
[image]
[image]

iceman1001, 2018-01-22:
yup.  Its the wmi selection that fails.   @bogiton  is doing a fix for it. :)

kevin2008-01, 2018-01-22:
[image]

iceman1001, 2018-01-22:
ok.  restart the client, I've deployed a new version

kevin2008-01, 2018-01-22:
Goooalllllllll ++++++++

kevin2008-01, 2018-01-22:
[image]

iceman1001, 2018-01-22:
Restart once more and the text looks better. 

I suppose this means v1.0 release is due

kevin2008-01, 2018-01-22:
oh very good !!!
Please don't loose sak return (-.-) fonction.
V1.0 béta is valided....
Thank you @iceman1001 @bogiton

iceman1001, 2018-01-22:
I updated the readme.md file,   this issue will be closed.

closed 2018-01-22 by iceman1001