iceman1001/ChameleonMini-rebootedGUI #15: bin diff functionality in GUI

iceman1001, 2018-02-04:
We mentioned some ideas of a tab with bin-diff functionality..

Just saw this one,  which have more than 2 object diffs... which would be interesting for us.
Don't ask if the licens or the rest fits..   Just saw it.

https://github.com/juhakivekas/multidiff

bogiton, 2018-02-05:
I like the idea of multiple hex editors. With our current GUI layout we can easily have up to 4 difference hex edit controls visible at the same time. I was playing around a bit with that Be.HexEditor control, which seems very fitting, but I am still trying to figure out a user-friendly way to open/save the different editors. The easiest way would be to have separate open/save buttons for each one.

iceman1001, 2018-02-05:
How about we solve one problem at a time?  Get some hex-editors in, is step one.   with multiple buttons.  
and step2 would be us figuring out a easy gui given what we have and want to do.   :)

bogiton, 2018-02-05:
Hehe you are right. Ok, I will add/test/share them soon.

bogiton, 2018-02-05:
Really temp. Let me know what you think:
https://github.com/bogiton/ChameleonMini-rebootedGUI

iceman1001, 2018-02-05:
nice!   I think four is too many.
Maybe three at the most.  
And we would need to make them  "dockable"  MDI..   so we can adapt between 16bytes width vs 4 byte width..

bogiton, 2018-02-06:
I was testing out the open source dockpanelsuite and it turns out to be a very good alternative of the paid ones.
I am attaching a mockup I made, while testing it.

[image]

Of course, in order to make this functional, it will require some (or lots of) additional work but I think that the result will be quite appealing.

iceman1001, 2018-02-06:
indeed,  you took the idea to another level.  I was thinking inside the diff-tab,  to have a MDI window there..

It looks nice but from a usability perspective,  i would want to see dumps side by side with highlighted colors to show the differences.

bogiton, 2018-02-06:
Hmm yeah, that would be hard for separate forms/dockables. As an alternative we could provide some options for the byte-width and resize the controls accordingly.

iceman1001, 2018-02-06:
There is three **major** widths,   4, 8, 16 bytes.     We should be able to identify between them quite well.
otherwise just a radio button away ;)

But you agree looking at dumps is something to be done side-by-side...

bogiton, 2018-02-06:
Yes, I agree that these have to be side by side. I am just considering the option to have a separate window for this tool and not be part of another tab. That way, we can display a dialog when someone wants to bin-diff dumps, like the following:
[image]
In that dialog we can have the user select 2 or 3 different files, along with the desired byte width, and then dynamically create the new window for it.
What do you think?

iceman1001, 2018-02-06:
Sure,  why not a separate form for it?    But that popup looks bad,   I rather have drop-n-drag,  and trying get as much in a easy flow.   with detection of dumps size,   comparison is bytewise,   always highlight with colors,  ..   
Like:   select three files,  drop them on the form,  and smack loaded...

iceman1001, 2018-02-18:
@bogiton  how is the hexboxes going?  _curious_

bogiton, 2018-02-18:
I had that postponed for a while, but I will get back to it soon. I already have tested the drag 'n drop feature, I just need to combine everything nicely before sharing :)

iceman1001, 2018-02-18:
don't worry,   it never has to be perfect before commiting :)   I'm ok with work in progress..  
Thats why we have **releases**....

bogiton, 2018-02-20:
It seems that maybe the choice of HexBox control was not the best one. It doesn't provide direct access to each byte's component properties, like fore color. This means that in order to color-code or highlight the different individual bytes among the HexBoxes would require changes in the source code of that control. Do you think we are up for that, or we should find an alternative?
On the other hand, I did include the drag 'n drop feature and the resize of the HexBox controls upon the different byte width, but I could only fit in 2 HexBoxes. You can check it out on my fork (https://github.com/bogiton/ChameleonMini-rebootedGUI).

iceman1001, 2018-02-20:
hm..  the coloring is a must, as is drop-'d-drag.
Time to look for different control,  or you can open an issue for hexboxes repo and ask about it?

bogiton, 2018-02-20:
The project is located in sourceforge, but I think the developer has really abandoned it (last version is 2013-12-27). There are couple of users who request the highlighting feature, like this one (which includes the suggested steps by the developer to achieve that): https://sourceforge.net/p/hexbox/discussion/695598/thread/8f158493/
And while searching, I came across to this guy who apparently has tweaked the HexBox control to support highlighting for specific bytes (instructions in his case): https://github.com/pmsanford/sic-debug/commit/860fa5520ba3274c6a3d71e127f8fbcdf3c296ad
Maybe we could give it a try. What do you think?

iceman1001, 2018-02-20:
Hey,  2013,  then its stale.  Its worth a try, go for the pmsanford's fix!

iceman1001, 2018-02-24:
@bogiton   how is the bin diffs going?

bogiton, 2018-02-26:
Sorry for the delay. Well, it kinda works now. At least in a use case like this:
- Drag n drop 2 files anywhere in the "Dump Management" tab to load 'em up in the two hexboxes
- Press "Compare" to highlight the differences
- Change any byte in any hexbox and then press "Compare" again to update the highlighting

Check it out if you want: https://github.com/bogiton/ChameleonMini-rebootedGUI

iceman1001, 2018-02-26:
Nice.
I think we can have auto-compare on..   and ontextchange event to trigger the compare-update aswell.

bogiton, 2018-02-26:
Yes, I was thinking the same thing. Actually, I was looking for an event to handle when a byte is being edited in order to force comparison on that byte only, but couldn't find one. Maybe I should just tweak the PreProcessWmKeyDown method.
Also, do you like the drag-drop feature on the entire tab, or should we just allow it for each hexbox?

iceman1001, 2018-02-26:
Yeah,    drop-drag event on the winform / tab..    tab is better..
There are a lot of events for keys  keystart, pressed,  etc   try finding one,  like keyup

bogiton, 2018-02-27:
I ended up raising an event from the WriteByte method of the ByteProvider to the HexBox and from the HexBox to our form, where it's handled. It still updates the whole thing upon editing a byte, but it works. Let me know what you think.

iceman1001, 2018-02-28:
look at that.  Nice!
I would prefer it was the background that changed instead of text color.
and for the 16byte layout
that we made the boxes fit 16bytes side by side,    we make main form wider.
and we keep sizes of boxes same.   
but we allow the 4 / 8 / 16..   

what do you think also about,   
if filesize is  1024, 2048, 4096, 8192,   (what mifare classic mini size?)  then we autoselect 16bytes.

if  smaller than 256,  we go for 4bytes.

bogiton, 2018-03-01:
I like your ideas. And they are very doable.
About the autoselection of the byte-width; what if the files differ in size?
Also what about the 8-byte width? No autoselection for that based on a file size?

iceman1001, 2018-03-01:
iClass uses 8-byte width,   and that aint happening on chameleon mini in awhile.  I dont have a clear case for 8byte widths based on size..  iclass is 2048, 32698...  so no.  

Good catch,  filesize difference,  should be noticed in a box or something,  indicating the different sizes.
So two groupboxes around hex-box,   where we write name and filesize in.
and the indicator field somewhere stating "differnt sizes"

bogiton, 2018-03-02:
It looks much better now. The changes we discussed are already on my fork.

iceman1001, 2018-03-02:
Much better!
the drop and drag could be more intelligent,    it work perfect when I drop 2files,
but for 1 file at the time,  it insists on load the first box over and over when I drop.

iceman1001, 2018-03-02:
@bogiton  If you make PR,   I can start adding the multi-language support

bogiton, 2018-03-02:
I fixed the drag n drop a bit, so that it checks where the drop is happening and loads the file to the corresponding hexbox. I will do PR now.

iceman1001, 2018-03-02:
*merged*   awesome!

iceman1001, 2018-03-02:
Adapted it a bit.   I guess this is how I would like it.
I wonder if there is possible to get ascii column aswell.   
Next would be different background color for the bytes you edit manually...  to show difference..
@bogiton   what do you think?

iceman1001, 2018-03-02:
yes!  the ascii is already implemented in the control,  property   **StringViewVisible** == true  :)
And the control tells you the needed minWidth with  property **RequiredWidth**    _perfect_

bogiton, 2018-03-02:
I was just typing exactly that :P
This would show up nice in 4 and 8 byte widths, but it won't fit into the 16-byte one.

iceman1001, 2018-03-02:
Indeed, we are limited with space.  Well we can always increase the main form.  
or let the Output box be collapsable into the right.
We are free to fiddle with the GUI as we please :)

bogiton, 2018-03-02:
Hmm.. Actually, now we could put the output box to the bottom of the screen and use its former space for the controls, without making the main form too wide.

iceman1001, 2018-03-02:
yup,  since we need the height in the form to look better at dumps

iceman1001, 2018-03-06:
@bogiton   soon enough you have built all the needed tools to collect those pescy uid-samples needed...

bogiton, 2018-03-06:
Hehe yeah, let's hope so.
Btw, where are we with the bin-diff functionality? Is there something else missing?

iceman1001, 2018-03-06:
well, we still have the idea of moving it to the bottom you mentioned.
I'm swamped with stuff so no time over for this project...

bogiton, 2018-03-06:
Ah right! Ok, I will work on it during the week.

iceman1001, 2018-03-16:
@bogiton   did you find time to finish it?

bogiton, 2018-03-16:
I have already made a PR with the relocation of the status textbox in the bottom of the screen, which you have accepted. Is it incomplete?

iceman1001, 2018-03-16:
:)   ...totally forgot that.   Too busy.  
and the last PR fixed up the project,  Thanks!

The colors looks ok,   the width,  the ascii part,    nice!

What do you say about  _synced_ scrolling?    Making a checkbox to make it optional,  but once checked, it locks scroll,  so pulling on one side,  also scrolls the other...

bogiton, 2018-03-16:
Heh, I thought about synced scrolling too, but I was worried about handling different file lengths. The checkbox idea is much better and safer. I will try it out.

iceman1001, 2018-03-16:
I pushed out a release of this version.  [1.0.0.8]
And closing this thread.   
We can start new one

closed 2018-03-16 by iceman1001