iceman1001/ChameleonMini-rebootedGUI #17: enhancements to GUI

iceman1001, 2018-02-24:
@gtpy  has used and enhanced our ideas for RevG,  and some of which we should adapt.
Like the idea of showing a picture of the identifyed device.  That would be so easy and still so good.

ref:
https://github.com/gtpy/ChameleonMini-RevG-GUI

gtpy, 2018-02-24:
I tested your GUI with Rev G but it doesn’t work. There are some problems with not identified commands. I’ll analyze it later and do a PR.

iceman1001, 2018-02-24:
Since I don't have RevG ,  its good that someone who has it can test for us!  
Looking forward to see the PR!

iceman1001, 2018-03-06:
@gtpy   have you found time to find the cause of the problems?

gtpy, 2018-03-06:
Not yet.
Really much to do in RL.
I think I can look at it till end of this week.

Sent from ProtonMail Mobile

iceman1001, 2018-03-16:
@bogiton 
Started to add the picture identifying on tab "settings"     Two pics  RevG and RevE rebooted so far.

Also added context menu to toggle sync scrolling.
would need to add it to the hexbox's content menus..

bogiton, 2018-03-17:
Pics are a nice addition, indeed! Of course, it is identifying my customized firmware (removed the "MY" extension) wrongly as a RevG, so in regard to future firmware updates maybe we have to use the PID&VID for identifying instead.
Yeah, I just saw the context menu item, cool! I will also added to the hexboxes today.

iceman1001, 2018-03-17:
The firmware name should be modified for the next releases :)
and a better detection between official RevG ...

The contextmenu is great..    but I like the addition of "mouseenter"  which focuses the hexbox when over it,  enableing to scroll directly with mouse-scroll  :)

bogiton, 2018-03-17:
Ah right, I haven't had that problem, cause I'm using Win10, which does that automatically. But yeah, extremely useful feature. :)

iceman1001, 2018-03-17:
hm, I have that on Win10,.. 
Try loading a dump  and "unselect" the hexbox by clicking on the log textbox,    move mouse to hexbox and it doesn't scroll if you deactivate it in the source. In any case,  its much more easier to use with it :)    The way I like it.

Maybe some borders around the image...

What do you think about the idea of color-templating?    So we make a color template for Mifare Classic S50,  S20, S70,..   or  UL..  Selectable from a dropdown?

bogiton, 2018-03-17:
Selection of back/fore colors sounds like a nice touch. Maybe colorpickers would be more user-friendly. We just have to add that twice, one pair for each hexbox.
We could put the image inside the "connection status" group box and re-arrange the whole layout of the "Settings" tab a bit.

iceman1001, 2018-03-17:
No, not picking colors,  I am thinking on colorizing the memory layouts..

Like this:
[image]

Don't let me stopping you fixing the settings tab :)

bogiton, 2018-03-17:
Ah ok, yeah that looks interesting. I'm worried though on how would they pair with the the diff colors we have.
After the hexbox context menu, I'll give the settings tab a try!

iceman1001, 2018-03-17:
well,  different modes?  template mode vs compare mode?   
I start a new thread for it.

looking forward to see the context menu,   you will need to expose a  event and expose the hexbox contentxmeny control property to be able to add a menuitem dynamically

bogiton, 2018-03-17:
Yes, exactly. I have it ready actually, I will PR in a few.
Will be fitting, for the Chameleon to switch colors :P

iceman1001, 2018-03-17:
LOL
I'm closing this one since we have the picture identification.

closed 2018-03-17 by iceman1001