iceman1001/ChameleonMini-rebootedGUI #18: multi language support

iceman1001, 2018-03-01:
Adding multilanguage support. 
I think that textfiles with different languages will be fine.
and a dropdown to select language.
also adding saving the prefered language in the config file.

iceman1001, 2018-03-21:
@bogiton    I just added the multilanguage support.
Would you mind testing it out.   I know the hexbox isn't translated since I only extracted control.Text values into the language files

bogiton, 2018-03-21:
Ha! This is very cool! Never tried the multilingual capabilities of .NET before. There are some strings missing from the translation file, as you said, but it works perfectly nonetheless! Good job! :)

iceman1001, 2018-03-21:
its actually not the builtin globalization support in dot.net.   That involves  compiled localized dlls which can't be edited later on. 

Its my own take on it.  With simple text files. 

I asked for a Chinese translation.  Dunno if you want to add a greek one ;)

closed 2018-03-21 by iceman1001