iceman1001/ChameleonMini-rebootedGUI #24: Color templateing of card memory

iceman1001, 2018-03-17:
Color templating of card memory,  in order to see what a card structure looks like.
two modes?  compare and templated since the colors might mix..

should it be textfiles for easy editing?

bogiton, 2018-03-18:
The coloring would apply to the different sections of a dump, like config area, user data area, etc, based on the card type? If so, the byte ranges are pre-defined, right?

iceman1001, 2018-03-18:
Yes,  like UID bytes, BCC, SectorTrailer on MifareClassic.  (or divide it to keyA,  keyB,  access)

if we only use byte ranges,  or following columns in a csv file,
startindex, len,  color,   #comment,
if would allow for easy parsing.

iceman1001, 2018-03-21:
I started with this one.
thinking of  a simple CSV file format with the following fields.

startposition; length (incl pos);  background color; foreground color;  description text

#  denotate a comment and line will be ignore.

iceman1001, 2018-03-21:
First try at templates.  :)

[image]

iceman1001, 2018-03-21:
@bogiton   Would you mind looking for bugs?

https://github.com/iceman1001/ChameleonMini-rebootedGUI/commit/3d6ee2951429f0719ba83cead99699bf0ed3a3e9

bogiton, 2018-03-22:
Looks very promising!
One minor bug, where the template colors weren't get removed after selecting the "Not selected" item from the the dropdown when a single dump was loaded. I did a PR along with some extras!
Also, maybe we need to add something like a pop-up with a color legend for the templates.

iceman1001, 2018-03-22:
the "not selected",   as it works today.

- when you have two files loaded,   it does the compare again.  (remove template but show diff)
- when you have one file,  it doesn't remove the template..

I was thinking on a legend,  thats why I added the description field in the template file.  :)

You didn't notice the fab possibility to have startpos and length fields as _expressions_ ?

not just fixed values..

iceman1001, 2018-03-22:
Here is a color table with knowncolor names:
http://www.flounder.com/csharp_color_table.htm

I updated the selected template when loading a file.

iceman1001, 2018-03-22:
Alrigthy then,    sorting order on templates and languages.  Move the control.

People can add their own templates now.

closed 2018-03-22 by iceman1001