iceman1001/ChameleonMini-rebootedGUI #26: GUI draft for release v1.1

iceman1001, 2018-03-18:
@bogiton   I started with a draft,  feels like this current status is good enough for a new release.
Would you mind read and adapt it?

bogiton, 2018-03-18:
I did edit it a bit. I will also try to re-arrange the settings tab today, as we discussed. Maybe we should change the recognition of the devices based on the PID&VID? After these, I believe we will have a nice new release.

iceman1001, 2018-03-18:
And we are going to start with making a branch for each release.

iceman1001, 2018-03-21:
So,  I tried to add the VID/PID for RevG.   The picture was wrong when no device was detected.
Not sure if its the right VID/PID.   And someone said RevG not working on the GUI..

1. waiting for chinese languatranslation

What more would we need???

bogiton, 2018-03-21:
The VID/PID for RevG seems correct. I double-checked it in other repos. But the string to compare it with needs to be like this: "VID_16D0&PID_04B2".
I will provide a greek translation too, but some controls need to be resized a bit to fit the translated text. So, I will also do that later tonight.
After that: Happy new release! :)

iceman1001, 2018-03-21:
I dunno if anyone besides English and Chinese will need a translation.

iceman1001, 2018-03-21:
So this release will also get templates for Memory dumps  :)

iceman1001, 2018-03-22:
Got the chinese translation. 

Time for release  @bogiton ?

bogiton, 2018-03-22:
Almost there! I'm doing some minor resizing again to fit all the translations. Will PR shortly.

iceman1001, 2018-03-22:
:)
Some mistake,   the **open location** button is not supposed to open languages folder.
I just put it in the languages groupbox,  since it didn't it in otherwere.

I suggest we change the groupbox name for it to be  Languages & Templates

bogiton, 2018-03-22:
Ah, ok, sorry. Yeah, I thought it was meant to open only the languages folder. Yeah, we should rename the groupbox to "Languages & Templates", as you suggest.

iceman1001, 2018-03-22:
no worries,  it was I, who was unclear.

bogiton, 2018-03-22:
Should I close the PR, make the changes and do a new one?

iceman1001, 2018-03-22:
do what will be easiest for you.

bogiton, 2018-03-22:
I did the changes and the PR was updated automatically! Magic!
Should be ok to merge now.

iceman1001, 2018-03-26:
well,  we got franch translation and I'm expecting a italian soon.  Also the serial tab is there,  might wait with the release until its finished.

kevin2008-01, 2018-03-27:
iceman don't loose +1 / -1 increment. thank

iceman1001, 2018-03-27:
+1 / -1 ??  in the GUI?   I'm not following,  please explain.

kevin2008-01, 2018-03-27:
all is ok in last update. left and right increment...

iceman1001, 2018-03-27:
I don't think RevE firmware has increament.   
if you use a RevG ,  those inc - modes should show up,  since we query the device which supported modes it has.

kevin2008-01, 2018-03-27:
serial interface working ? actually not for me rev e rebooted.

iceman1001, 2018-03-27:
How does it look when you send?

I just pushed a fix for the send button, which didn't send anything.
Now you can send via button,  or  with pressing enter

iceman1001, 2018-03-27:
The serial interface tab is really not for amateurs.  
The GUI itself also assumes people know of what a chameleon can do or does.

iceman1001, 2018-03-28:
With translations, color legend, dump management, serial interface,  I think its ready for a release.

Some questions asked in other threads by @kevin2008-01 
a) tab serial, add "?" to help text
b) tab operations,    +1 / -1 increment uid 

When it comes to a),   its not useful since the full commands uses  ? for query,  = for set, 
when it comes to b) ,  I think inc/dec uid is a firmware mode issue,  not a GUI issue.

kevin2008-01, 2018-03-28:
right. wooo

(a)For "?" is just in serial interface, in description command add "versionmy?" at "versionmy", and many command, it's simply for copy paste for newbies. Yes for"?", if not, command not working.
I explain, in command description, i see "command versionmy"  when i send command i write "versionmy?" for working, if no "?", no working, understand me?

(b) +1 and -1 is left and right increment or decrement, but it's implemented at last version. but i'have E rev, and is instead, but not tested yet.

(c) in upgrade firmware, please, precise, upgrade with iceman firmware directly for newsbees. 

For sak 08/88, i have time, i explain to you in future.

Thank's and see you.

kevin2008-01, 2018-03-28:
now i see.
good job. the result is perfect.

iceman1001, 2018-03-29:
A) Ii don't target newbies with this GUI.  its for medium/experienced users.   If you don't understand the commandset nor read the differences of revE / revG firmware you should not use the serial interface.  I don't even want to use it.  Its usage makes even the pm3 client look easy.
For newbies you have the  Tab  Operations in order to use the chameleon.

B) if the inc/dec  modes are active inside firmware,  they should show in the dropdowns on  tab Operations.

C) sak/atqa needs firmware change,  until then there is no reason to change GUI.

--- All questions solved.
its time for a release.  @bogiton  do you want to change something in the description of the release?
B)

bogiton, 2018-03-29:
I did edit it a little bit. I believe we are ready to rock :)

iceman1001, 2018-03-29:
And now I got the italian translation from @asper   (thanks bro!)   
Its ready to be released

iceman1001, 2018-03-29:
Its a release!
Thank you to all contributors!   (@bogiton , @kevin2008-01  @asper  @proxgrind)  
We made a good run this time.

closed 2018-03-29 by iceman1001

kevin2008-01, 2018-03-29:
if, you need translator for pm4, i'm here... I'm, in the road for july.... congratulations.....