iceman1001/ChameleonMini-rebootedGUI #30: GUI - tab for serial interface

iceman1001, 2018-03-22:
Suggestion,  I saw wolfgangmau s GUI has a serial interface which is a great idea.

GUI,
 - Tab for serial interface.

bogiton, 2018-03-22:
I would love a serial interface!

iceman1001, 2018-03-22:
I can start it after we merge your PR..

iceman1001, 2018-03-26:
@bogiton   I have added a Serial - tab.   Tell me what you think

bogiton, 2018-03-26:
I works perfectly! Good job man!

The only suggestions I have are the following:
- Add also a "Send" button for the sending of the commands
- Clear the command-send textbox after sending
- Get the available commands for each Chameleon version by sending HELP/HELPMY on connect

Again, great addition! :)

iceman1001, 2018-03-27:
...now, why didn't I think on getting the available commands via the help?  That is a spendid idea.

iceman1001, 2018-03-27:
@bogiton   
I adapted the following fixes,   
- clear button also empty the commandtext.
- help texts now loaded from firmware command and is selectable.

iceman1001, 2018-03-27:
and a send button.  
time to close

closed 2018-03-27 by iceman1001