iceman1001/ChameleonMini-rebootedGUI #32: color legend for templates

iceman1001, 2018-03-23:
as mentioned before,  to the new color templates it would be nice to see a popup color legend,  with the descriptions already inside the template files.

kevin2008-01, 2018-03-27:
in serial interface screen, add "?" after all command. VERSIONMY? .... Simply for copy/paste command. thank

iceman1001, 2018-03-28:
@kevin2008-01   What has your request to do with this issue?

iceman1001, 2018-03-28:
@bogiton   I added a draft for color legend,  well it works but it needs some love with the autosizing..

bogiton, 2018-03-28:
Very very nice! The only issue I noticed is the movement of the last legend item to the right. Ah and also, the legend doesn't get emptied when "no template" is selected. I will make a PR for them asap.

iceman1001, 2018-03-28:
yeah, I know.  I just got frustrated with it....