iceman1001/ChameleonMini-rebootedGUI #39: eml format upload directecly

kevin2008-01, 2018-03-28:
eml format upload dump directecly,  is possible in future ?

see you

iceman1001, 2018-03-29:
the concepts of eml file is counter to what a hex-edit does.
you should be able to just drop and drag a eml file already, but you will see why its not fitted for a hexeditor.

How about some converters?  That would be nice,  
EML -> BIN  
BIN -> EML

kevin2008-01, 2018-03-29:
Yep, converter is fine...

iceman1001, 2018-05-15:
@bogiton   What do you think of this style for converting?

[image]

bogiton, 2018-05-15:
Hmm, I was thinking something like auto-converting when uploading to a slot or when loading a dump for hex editing. But if we are gonna implement a file manager we could add more features to the context menu, like add to hexeditorX or even load directly to slotX.
Another option could be a "Converter" button in the dump management tab (since converting seems like a part of that) that displays a dialog to select the files to convert from/to and the type of the conversion to perform.

iceman1001, 2018-05-15:
hehe,  I wanted first just a list of files with a checkbox  and try to work in batches..
but then I realised the dump files will be anywhere on the computer drives.
Also thought of button, but samething there, I might not want to convert the file I'm watching.

With a simpler filemanager,  we can add more possibilities like you say with "compare 1" / "compare 2"..
and yes,  upload to slot,   and we don't have to have all files in one place etc..

iceman1001, 2018-05-15:
In my first try I called the tab "Convert"..  I realised "Utils" would be better.

bogiton, 2018-05-15:
Yeah, let's go with the file manager then and we can add the other goodies there :)

iceman1001, 2018-05-15:
ok,  I pushed the code so far.  Missing some nice folder icons etc..  and layout etc..  but it can convert :)

bogiton, 2018-05-15:
You are damn fast! You even created your own file manager control?? We could have used a free one :P
Btw, I'm getting some build errors, about the following missing files:
Error: Could not find file 'Templates\LegoDimension.txt'
Error: Could not find file 'Templates\Ultralight-EV1-48b.txt'
Error: Could not find file 'Templates\VingCard EV1-48b  (pm3).txt'

iceman1001, 2018-05-15:
aha,  I see,  they can be excluded from project.  I pushed a fix for it.

Well,  c# is easy sometimes.  No need to over-do things.  There are blogs on internet to follow.  :) 
Now we need some image icons,  like folder,  locked-folder,  file,  drive.   four of them I think.

iceman1001, 2018-05-15:
I didn't want to make a full fledge file explorer, just something that fits our needs to work better.

iceman1001, 2018-05-15:
the icons should be added to the imagelist control..

hiwanz, 2018-09-15:
@iceman1001 It's better to create a project with this repo, so that contributors can follow the progress and do what they can do for the repo.

[image]

iceman1001, 2018-09-16:
@hiwanz   not the best place to chat about project.   I have however enabled it now.

hiwanz, 2018-09-16:
project is not for discussing things but planning them

hiwanz, 2018-09-16:
for example, you can add todos to the list, something like 

- [ ] Add "Send to hexbox" menu item to the context menu
- [ ] Need more templates
- [ ] "To MFD" need to be done
...

so that contributors can pick up what they can do to help

iceman1001, 2018-09-16:
This issue is not for discuss project.  This issue is about EML converting.  Keep issues clean and to the point.  
Start a new issue instead.

iceman1001, 2018-12-04:
https://github.com/iceman1001/ChameleonMini-rebootedGUI/projects/1

kgamecarter, 2018-12-05:
mfd format is same to bin. Just difference file extension name.
I think just remove the menu item.
Or implement change file name?

iceman1001, 2018-12-05:
remove in  "convert to..." 
but keep for open file..

closed 2018-12-09 by iceman1001