iceman1001/ChameleonMini-rebootedGUI #44: Templates for toy-to-life tokens

iceman1001, 2018-04-04:
The data mapping of several toy-to-life tokens like  Disney Infinity, Skylanders, Amiboo etc etc would be excellent to have color templates of.  

Thats why I ask the community for help.   A template file is extremly simple to do.

Sample
https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/master/ChameleonMiniGUI/Templates/Ultralight-EV1-164B.txt

kevin2008-01, 2018-04-10:
.Cyan Disney Infinity   Right, ? you need table?

iceman1001, 2018-04-10:
I would need a template file,   I have no time to make these files myself but I do have all data mapping from these different toys-to-life,   thanks for the suggestion.

kevin2008-01, 2018-04-10:
send the correct format you need for working. when i have moment i work for community.

kevin2008-01, 2018-04-10:
create only one sample you need iceman1001

iceman1001, 2018-04-10:
The sample file is quite easy to understand.

Byte offset; length; text color ; background color; description 

The colors uses the known names, see the link in first post. 

There is already some template files, they make a great starting point. 
Just add it to the template folder and you will be able to test it on a dump file direct

hiwanz, 2018-09-15:
[image]

I've add a new series templates, you will see it in next release or you can build it from the source by yourself.

iceman1001, 2018-09-16:
Yes,  good work,  this issue is about a specific subset of templates.

iceman1001, 2019-03-17:
@bogiton   https://twitter.com/kaitai_io/status/1107011025222750208
Look at that fancy visualisation...  that would have been cool to have

bogiton, 2019-03-17:
Wow, yeah, this is cool! How could we integrate something like that? Create a logical tree structure of each data section?

iceman1001, 2019-03-17:
We don't need a structure to start with.  Keep it simlpe.   Like the template has rows,   we would just print those but when click on one, it would highlight the selected bytes :)

iceman1001, 2019-09-14:
Since no-one is working on this,  I will close it... 
Maybe @McEloff  has the energy to add it?

closed 2019-09-14 by iceman1001