iceman1001/ChameleonMini-rebootedGUI #53: Prerequisite problem

hiwanz, 2018-08-28:
the README says "prerequisite dot.net 4.6.1",but the DynamicExpresso.Core  in packages.config uses net462 as its targetFramework,see the code below:
[https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/32463ba953a9ddf9c78701a46bd6d4cc9d983944/ChameleonMiniGUI/packages.config#L3](https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/32463ba953a9ddf9c78701a46bd6d4cc9d983944/ChameleonMiniGUI/packages.config#L3)

I tested click-once deployment installation on Windows7,it turns out that net461 will fail the installation,net462 works fine.Is this expected?

iceman1001, 2018-08-28:
Great catch!   It was not expected.  Might need a fix for it.

hiwanz, 2018-08-29:
According to the documentation [https://github.com/davideicardi/DynamicExpresso](https://github.com/davideicardi/DynamicExpresso),  Dynamic Expresso Available platforms: .NET Core 2.0, .NET 4.6.1

iceman1001, 2018-08-29:
The solution would be to increase the pre-requsite to v4.6.2  for all parts.    Might even set 4.7.1 or higher.

closed 2018-08-31 by hiwanz