iceman1001/ChameleonMini-rebootedGUI #59: Converting function in UcExplorer

hiwanz, 2018-09-27:
Is "To Bin" the same with "To MFD", or is "To bin" means "to mfd binary", I didn't see the implement of "To MFD" menu item in UcExplorer.cs
```csharp
var emllabel = new ToolStripMenuItem { Text = "To EML" };
var binlabel = new ToolStripMenuItem { Text = "To BIN" };
var mfdlabel = new ToolStripMenuItem { Text = "To MFD" };
```

[https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/f43d4bbd5cc3da06764d8882bbfd5dd9d9b40fcf/ChameleonMiniGUI/UcExplorer.cs#L281](https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/f43d4bbd5cc3da06764d8882bbfd5dd9d9b40fcf/ChameleonMiniGUI/UcExplorer.cs#L281)

```csharp
switch (to)
{
    case ConvertFileEnum.Eml:
    {
        ...
    }
    case ConvertFileEnum.Bin:
    {
        ...
    }
}           
```

[https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/f43d4bbd5cc3da06764d8882bbfd5dd9d9b40fcf/ChameleonMiniGUI/UcExplorer.cs#L301](https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/f43d4bbd5cc3da06764d8882bbfd5dd9d9b40fcf/ChameleonMiniGUI/UcExplorer.cs#L301)

hiwanz, 2018-09-30:
According to [http://nfc-tools.org/index.php/Libnfc:nfc-mfclassic](http://nfc-tools.org/index.php/Libnfc:nfc-mfclassic), **MFD stands for MIFARE dump file**, the "Interpretation" section has explained the structure of .mfd file, which is totally the same with the .bin file, so "To MFD" could be removed, could I?

hiwanz, 2018-09-30:
OK. I'll deal with it, see it in next PR

closed 2018-09-30 by hiwanz

bogiton, 2018-10-01:
I think that the difference between MFD and BIN is that the first one represents a binary dump in "Chameleon" format (tag sectors first and extra info afterwards), while the BIN is the binary dump that comes from Proxmark (first the extra information and then the actual tag bytes).

iceman1001, 2018-10-01:
well.  it depends..  
binary pm3 files for MFU (ultralight etc) is different,    Dumps from mifare classic is just a memory dump binary.   The question is when is MFD format used?   only for mifare classic dumps?
Do MFD have extra info in it?

iceman1001, 2018-10-01:
..one of these days a json format should emerge...