iceman1001/ChameleonMini-rebootedGUI #62: Chameleon Rev G/ GUI issue

nico0481, 2018-11-16:
Hello,

I'm trying to use Chameleon Mini Rev.G with GUI, I'm not able to make it working.
I use last GUI v1.101 and Chameleon Mini is also uptodate.(same problem with previous version of GUI)
Chameleon Mini is well recognized by GUI and driver is installed (Chameleon Mini Virtual COM Port in device manager)
I'm able to send commands trough simple terminal and also via GUI
Every textbox in the GUI show (200:UNKNOWN COMMAND), as soon as I try to change parameters I get error message.
On Windows 10

How can I fix it?

Thank you

Best regards

Nicolas

gtpy, 2018-11-17:
@nico0481 I also recognized this.
Till now I hadn’t enough time to take a closer look at it.
I’ll fork it and take look at it. Should find some time next two weeks.
If you are familar with the commands of Rev G you could take a look at the code too.

gtpy, 2018-11-17:
@nico0481 if you only use RevG board you can use my GUI which only support the RevG.
If you use both Rebooted and G then this one is a good choice.

gtpy, 2018-11-17:
Damn it, I don't have much time right now.
Code cross-checked and determined that e.g. with the buttons only the RevE is queried. Because we uss L+R buttons. 
The terminal commands are also different. But I will compare them with your list on occasion.

Vrumfondel, 2018-11-29:
I'm currently in the same situation, but found a couple of issues already which need to be changed (and I did allready in my local clone of the code).

Most important (and took me a while to figure out): the size of rx-data needs to be increased - the original 275 bytes are exceeded by at least the response to HELP. This causes a lot of weird things allready, because the rest of the response is no dropped but added to the next response read from the serial port.

gtpy: How would you generally prefer - listing all issues within this issue, or opening a new one for each item?

What do you think of co-development, ones finished I could create a pull request for my changes.

gtpy, 2018-11-29:
I think as long as it is an issue with the RevG, we can bring these here together.
As I already wrote, I am currently very busy at work. 
I'm not sure when I can help out here.

Create PR as much as necessary.
I'll do the same when I get to it. :)

iceman1001, 2018-11-30:
A nice PR ,

iceman1001, 2018-12-04:
With latest PR  https://github.com/iceman1001/ChameleonMini-rebootedGUI/pull/75  this issue will be closed.

closed 2018-12-04 by iceman1001