iceman1001/ChameleonMini-rebootedGUI #67: todo: Add support for new json dump format

iceman1001, 2018-11-29:
**Todo**
Add support for new json dump format just introduced in proxmark3 RRG repo.

kgamecarter, 2018-12-01:
Is there a plan to use the [Newtonsoft.Json](https://github.com/JamesNK/Newtonsoft.Json)?

iceman1001, 2018-12-01:
Dunno which json implementation yet,  but there are minimalistic fast ones.

kgamecarter, 2018-12-02:
Consider DataContractJsonSerializer. It's not fastest one, but built-in .NET Framework.
Is there a JSON sample file?

iceman1001, 2018-12-02:
..I haven't actually tested the new dump format,   so hold on....

And here it is,  
https://pastebin.com/x6Xwc1EW

iceman1001, 2018-12-02:
fastJSON is one that I have used earlier but it always comes down to how to read nodes inside it in a nice easy way...

bogiton, 2018-12-02:
Hmm, sector keys are saved in the dump? Until now, for the RevE we only save the blocks of the tag.

iceman1001, 2018-12-02:
yeah,  I see it as optional.  the data part is most important.

iceman1001, 2018-12-02:
we add the keys in the sector trailer as in pm3 binaries or eml.   You can still do that,  but we also extract them in json.

kgamecarter, 2018-12-02:
I finished upload json dump.

Hexbox needs to be implemented ApplyChange.
Also need to write back to the sector trailer?

iceman1001, 2018-12-02:
Sadly you will have to rebase it ...  I merged the old PR...

iceman1001, 2018-12-02:
Not sure with the hexboxes,   I will need to see your code first...

kgamecarter, 2018-12-02:
Here is upload dump https://github.com/kgamecarter/ChameleonMini-rebootedGUI/commit/384e13f0daf155d75849409feeb847cf6602052c

iceman1001, 2018-12-02:
Cool,   yeah  matching the sectorkeys would be nice...
and reading of EML files aswell..

iceman1001, 2018-12-02:
The JSON addition seem to work.  I can drag and drop a json file onto the dump management tab and it loads.  When saving the json swapps a bit around the different parts,  so that can be confusing the first time.  However the change in block data is correct so far.  Also sector keys is updated when a Sector trailer is modified.  How cool is that!  :)

iceman1001, 2018-12-02:
ok,   [DataMember(Order=1)]  sorted those things out.   The outcome json file is identical as PM3 style.  :)

closed 2018-12-02 by iceman1001

iceman1001, 2018-12-02:
Now its the upload/download..   I noticed the download tried to save it as 4k

iceman1001, 2018-12-03:
it should but the  RevE/RevG PR  messed up the load of tag slot settings for RevE  (which I am testing with)..
So för RevE the GUI doesn't work right now :(

iceman1001, 2018-12-03:
the 4K format has a different sized sectors,  which isn't dealt with in current solution.

kgamecarter, 2018-12-03:
Yes, i need the 4k json sample

iceman1001, 2018-12-03:
no,   no Mifare Ultralight / NTAG json exists yet.

kgamecarter, 2018-12-03:
I tried to make 4k json.
please check 32~39 sectorkeys are right format.
https://pastebin.com/wdRZ7KiX

RfidResearchGroup, 2018-12-03:
The csave should be able to do 4k.  I just dont have access to a gen1a 4k card right now.