iceman1001/ChameleonMini-rebootedGUI #73: Not connecting to Chameleon any more

Vrumfondel, 2018-12-04:
Somthing got broken in establishing the connection;
On 1st look it seems the GUI tries to connect to the first available COM-Port, no matter what this is ..

iceman1001, 2018-12-04:
Not really,  it will try getting which serial ports is available,  then it tries to identify if the current port has a vid/pid connected to a chameleon.   if not, then loop to next available serial port. 
See method  https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/master/ChameleonMiniGUI/FrmMain.cs#L1207

Vrumfondel, 2018-12-04:
Im currently debugging this.
Currently it seems pb_device.Image doesn't like to load warning.ico ....

iceman1001, 2018-12-04:
Bugger,    We need a image for  "Unknown device" ...   Can't show RevE or RevG....

Vrumfondel, 2018-12-04:
No question about having a seperate image there, I just say that the current image-type seems to be the wrong one ...

Vrumfondel, 2018-12-04:
Confirmed: An icon can not be casted into an bitmap ....
Converting the .ico to an png and using this as ressource works again ...

iceman1001, 2018-12-04:
PR?

Vrumfondel, 2018-12-04:
Will come in ~ 2hrs ...

closed 2018-12-04 by iceman1001