iceman1001/ChameleonMini-rebootedGUI #84: SaveActiveSlot() saving wrong value

Vrumfondel, 2018-12-11:
Moving this to a own function is a good idea, but current implementation is wrong:
Doing a slotindex++ in SaveActiveSlot() is only working for RevE, since _tagslotIndexOffset is 1.
For RevG the offset is 0, so there is 1 added when saving, but not substracted when restoring --> Marked slot is one index to high ....

I will fix this in my next PR

iceman1001, 2018-12-11:
looking forward to the PR

iceman1001, 2018-12-12:
Yeah,   move to operations tag and show result in output window,   and become disabled on RevE rebooted firmware..

iceman1001, 2018-12-12:
So I moved it and resized the utils controls.  Looks better.
Now there is some translations missing because of new GUI

closed 2018-12-12 by iceman1001