iceman1001/ChameleonMini-rebootedGUI #87: draft v1.2 release

iceman1001, 2018-12-12:
@kgamecarter    
@Vrumfondel 
@bogiton 

I pusblished a v1.2.0.1 for those who use the click-once deployment. 
Also added a new branch in preperation for a new release.

would need some proper tests of  RevG functionality.   I have tested on my RevE rebooted.  It works as expected.

Is there any of the issuses still there,   (see issues and projects tabs)  which you guys maybe wanna fix?

Great work!

iceman1001, 2018-12-12:
and yeah,  we need to update the pictures in the readme.md...    They really are missing stuff :)

iceman1001, 2018-12-13:
Cool,  MCT dump format supported done!   Thanks to @kgamecarter !

bogiton, 2018-12-13:
You guys are on fire! I press the refresh button and I see tons of fixed/added stuff each day!
Very good job!

iceman1001, 2018-12-13:
The love feels soo good here.  The GUI gets some well deserved love!    Dropping bin/json/eml/mct files and the dump management tab just takes it  :)    Feels so good!

iceman1001, 2018-12-14:
Pushed out the v1.2.0.5  as click-once deployment.

iceman1001, 2018-12-14:
OK,  if you all test it out.  I belive the new dump formats works nice.  
Some more color templates would be nice to have. 

Latest click-once deployment is  v1.2.0.7

iceman1001, 2018-12-15:
Alrighty then.    added a new iClass template and fixed the template control to fill out the size.   Looks good now.

Latest click-once deployment is v1.2.0.8

iceman1001, 2018-12-15:
How is the RevG testing going?   I looked at RevG firmware,  they really uped the game.  Its looks great.  RevE rebooted is behind.

iceman1001, 2018-12-15:
And yeah,  rumours :)   There may be a RevG rebooted device.  What more should it have inside?

Vrumfondel, 2018-12-15:
I don't see any issues with RevG so far ...

iceman1001, 2018-12-15:
There is a lot of more commands in RevG firmware.

There some minor stuff under project 'enhance gui' but I like this GUI now.   Great addons!

iceman1001, 2018-12-17:
so since nobody has ideas or suggestions,  I think I will close this issue and release  V1.2   

@bogiton   do you wanna edit the draft release (its there already)

closed 2018-12-17 by iceman1001

bogiton, 2018-12-17:
Seems perfect :)