iceman1001/ChameleonMini-rebootedGUI #88: To implement xmodem download in the serial tab

iceman1001, 2018-12-12:
When sending download command in serial tab,  the client crashes.
Since it waits for xmodem download and thats not supported yet.

Vrumfondel, 2018-12-13:
Anybody on this one yet? Otherwise I will have a look tomorrow ...

iceman1001, 2018-12-13:
all yours, bro!

iceman1001, 2018-12-13:
I suggest printing the downloaded bytes as ascii-hex..  [00 AA BB CC]

Vrumfondel, 2018-12-14:
Hm. In my opinion the serial tab is for more "lowlevel" work, so just sending and receiving plain information is what I would expect (like to quickly check the content without the need afterwards to open the dump again)
If someone wants to save the data there is still the "Download Dump" function in tap "Operation".

bogiton, 2018-12-14:
It sure is low-level, but then again why would someone use the GUI for serial commands? :P
Let's say that the print of the actual bytes when we issue the download command would suffice. But what should happen with upload? Write the bytes one-by-one in the textbox? Those two commands should be complementary. And since it is part of a GUI, there should be some easiness in it.

iceman1001, 2018-12-14:
No popup.  We have **tab operations** for GUI communications with the device.  
The serial tab is for sending text commands,   and if you send download,  you don't want it to crash.
I suggest we return a hex-string (32bytes in a row)  and print in the output window.

Yes,  fixed sized font is needed.  I have also another idea with different colors from when sending vs receiving texts.   I will push it..

Vrumfondel, 2018-12-14:
@bogiton: I frequently use the serial tab instead of some serial-terminal. Having the available commands clickable makes it quite easy to use.

@iceman: Just sent the PR, currently the format is 8-byte wise, but changing it to 32 is surely an option. I mainly want to avoid the need for horizontal scrolling in this case. 
What about making this depending from the actual tb-width - at least in steps like 8, 16, 32 bytes?

iceman1001, 2018-12-14:
Regarding byte width,   doesn't current output textbox window fit 32hex with spaces in a row?  
We have space to make box larger,  now that the help text is to the left.

iceman1001, 2018-12-14:
Merged!
Looks better,   I will push some minor adaptations to it...

strange thing,  my helpmy  doesn't show the DETECTMY ...

bogiton, 2018-12-14:
I get what you say about the DOWNLOAD, but what about the UPLOAD cmd? :)
It also crashes the GUI.
@iceman1001 I get DETECTIONMY in mine. As far as I remember, there used to be an issue in the firmware, where the declaration of the command was below the other ones, so the helpmy wasn't returning it.

iceman1001, 2018-12-14:
I say we disable  UPLOAD in the serial tab.   I don't want that functionality in there.

Aha,  I may have the original RevE rebooted firmware installed.  :)

iceman1001, 2018-12-14:
ok,  I pushed a fix where UPLOAD is blocked in tab serial.
Also Tab serial prints the DECETIONMY output in hex string.
That will take care of some crashes.

bogiton, 2018-12-14:
That's great! It is really coming out very nicely :)
If only we could get those timings in the firmware fixed :P

closed 2018-12-14 by iceman1001