iceman1001/ChameleonMini-rebootedGUI #90: Add support for MCT tool 's dump format...

iceman1001, 2018-12-12:
The MCT tool has its own format,  it would be nice if the GUI could support it aswell

https://github.com/ikarus23/MifareClassicTool/issues/217

@kgamecarter    Maybe your magic touch?

kgamecarter, 2018-12-13:
Ok
MCT is a nice tool for mobile read/write card.
I also want to support the MCT format.

iceman1001, 2018-12-13:
@kgamecarter  Great addition!    Now I only need a MCT dump file to test/verify it with  ( @ikarus23   ?)

iceman1001, 2018-12-13:
Converting back and forward looks great.
...may add a refresh button,  right-click,   or after a convertion to update the file explorer...

iceman1001, 2018-12-13:
whoops..  converting a Ultralight bin dump to EML or MCT ..  will try to make a classic  formatted....

kgamecarter, 2018-12-13:
MCT is a tool only for classic , so just only classic format.

Eml convertion code is copied from https://github.com/iceman1001/ChameleonMini-rebootedGUI/blob/1a345a9ee73312745cc4908b50945ed7142cd455/ChameleonMiniGUI/UserControls/UcExplorer.cs#L301
Does Ultralight also have EML format?
If so. I need a Ultralight EML sample.

iceman1001, 2018-12-13:
:)  the UL/NTAG dumps are more complex in one sense.   I have bin for it,  the json is not done yet...
...if you don't want to add json in pm3 client for **hf mfu dump**

kgamecarter, 2018-12-13:
ok
it's 4 byte per line.

iceman1001, 2018-12-13:
...the old dumps was 4byte, but then we added a block of special data in the beginning of the dump file

iceman1001, 2018-12-13:
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/cmdhfmfu.c#L1856

kgamecarter, 2018-12-13:
The special data does not seem to match EML format.
needs a better format for mfu_dump_t

iceman1001, 2018-12-13:
I know...   hence the urge to get JSON for it :)

iceman1001, 2018-12-13:
Alrighty then.    **hf mfu dump** new json dump format in proxmark3 client.

https://pastebin.com/gn5wG3KM

iceman1001, 2018-12-14:
as of this moment I don't think the Chameleon uses the special data.  It really should though.
Better look in the firmware repo.  Or @bogiton , do you know ?

As of now,  I think it can only load / convert JSON format. not save it.

iceman1001, 2018-12-14:
How about we just set those special fields as empty?

bogiton, 2018-12-14:
The only extra thing we save are the counters for UL, right after the memory blocks. I had a check added for this in the DownloadAndSaveDump method.

iceman1001, 2018-12-14:
But do we have access to Version, Signature etc?

bogiton, 2018-12-14:
Nope, just the memory block bytes. Signature is hard-coded I think.

kgamecarter, 2018-12-14:
The binary dump only end of user memory?
No include lock bytes and configurations page?
[image]

iceman1001, 2018-12-14:
There are so many different variants of Ultralight / NTAG etc class tags so keeping track of each cards memory layout is hard. Not to mention locking mechanisms. The pm3 client dumps all blocks it can read.

iceman1001, 2018-12-14:
if pm3 client can figure out default pwd/pack,   it adds those for known tag formats.  Not all formats.