iceman1001/ChameleonMini-rebootedGUI #96: Feature: Read out Card

Vrumfondel, 2018-12-18:
Just to start a discussion here:
At least the RevG has commands for rawWrite/Read to the NFC card. In theory that should anable to read out the content of any card (considering a known key).

Wouldn't this be a nice - and unique - feature ?

iceman1001, 2018-12-18:
Sounds like a nice feature for v1.3  👍   

Opens up for all the key-file stuff,  lucklily we have json/eml support already.  This will enable a faster development.

- MCT keyfiles
- PM3 keyfiles

iceman1001, 2019-02-26:
@Vrumfondel  did you manage to get something done with this?

Vrumfondel, 2019-02-28:
Well, I'm struggeling to get the corresponding commands working - what ever command I send out to a card I always get a timeout response.
I'm in discussion with emsec, but their response time is quite slow ....

iceman1001, 2019-09-14:
ping

iceman1001, 2019-12-20:
We have an identify button now.  Which prints it in the log text box.  
I think we would need a new tab with the dedicated RevG  reader functionality for a nice user experience.