iceman1001/ChameleonMini-rebootedGUI #98: Converting dump file for RebootedGUI

dealaddict, 2019-02-06:
Is there any android app that makes dump files supported by RebootedGUI? I tried MIFARE++ Ultralight but it makes dump files in .txt format and despite playing around with them it wont upload properly. I got excited about the Ikarus tool but its for Classic only and I need something that supports Ultralight dumps. Or if there is even a way I can convert the .txt files into properly formatted dump files that is supported by RebootedGUI.
[1502[1].txt](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/2837761/1502.1.txt)
[1506[1].txt](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/2837764/1506.1.txt)

FYI I already bought the ACR122U but its on its way and may take a month so in the mean time need something to play with.

Thanks

P.S. Sorry Iceman for using the wrong repo, I have now posted this in the correct Repo.

bogiton, 2019-02-07:
Well, actually, you can just transform the txt to bin format using a tool like xxd:
`xxd -r -p 1502.txt 1502.bin`
and then try to load that bin into Chameleon.

dealaddict, 2019-02-07:
> Well, actually, you can just transform the txt to bin format using a tool like xxd:
> `xxd -r -p 1502.txt 1502.bin`
> and then try to load that bin into Chameleon.

I used the command in UnxUtils but it didnt do anything. I might be doing something wrong as I am not too savvy with unix. is there any particular program for Windows that i can do this? @bogiton

bogiton, 2019-02-08:
@dealaddict I suppose you could use any hex editor for Windows and just paste the contents of the text inside that. I personally use Frhed (http://frhed.sourceforge.net/), which has an "Import from hexdump" option that does exactly what you need (I just tested it).

dealaddict, 2019-02-08:
Thanks @bogiton

So I copy paste the text contents into the hex editor and then should I be able to  save the file as.bin or do I paste the contents into the rebooted gui dump management section?

bogiton, 2019-02-08:
You can't create a new file from the dump management of the Chameleon GUI. You can only edit existing files.
I believe the easiest thing to do is to paste the txt file contents into the hex editor of your choice, save that as a bin file and then upload it to the tag slot you want via the Chameleon GUI.

dealaddict, 2019-02-08:
@bogiton  Thanks man. Finally I made progress and was able to make a bin file by converting the txt file into hex file. However it doesnt seem to be making a 1x1 clone of my tag data. By that I mean it doesn't copy the URL record from my original dump into the chameleon.

Particularly this text 55132F2F67742F63703F34303337 that turns into this URL in hex format **//gt/cp?4037** my chameleon is only able to emulate the UID but not its contents

bogiton, 2019-02-08:
@dealaddict Can you share the dump file when you press the download button in Chameleon GUI, after having uploaded your converted hex file? Theoretically, those two should match.
What kind of reader are you testing it against? It identifies the UID (with the select_uid command) but it can't read the user memory (where the URL resides)?

dealaddict, 2019-02-09:
@bogiton  I am using NFC tools PRO and Ultralight++ apk to read and make text files of these. I then converted these text files to the bin file in the editor. 
Its even messing up my UID now and also not able to emulate the text inside which should be the above exampled URL's from my earlier post. 

Please see download link for zip for my original text file  @dump + converted bin file + dump file downloaded from Chameleon GUI.
[https://ufile.io/e4w9k](url)

bogiton, 2019-02-10:
@dealaddict If you view the files you uploaded with a hex editor/viewer you can see why there is a problem with emulating. You obviously pasted the contents of the text files as text and not as hex byte values.
I made two proper bin files for you to test.

[dumps.zip](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/2848726/dumps.zip)

dealaddict, 2019-02-10:
Wow @bogiton you sir are a life saver. The two files you gave me works perfect! However I still don't know how to paste them as hex bytes. I tried different ways of pasting in the fred hex editor but my files dont convert as yours. I tried using the paste button as well as ctrl v.

Eventually on one website https://hexed.it I am able to paste it correctly but when I save it it won't work. I suspect it's because the whole text is showing up in one line.

Once I figure out how you are pasting them and saving them exactly i should be able to do it. 

LOL sorry for taking so much time of yours I'm still learning the ropes

bogiton, 2019-02-10:
I just tried out the https://hexed.it site (cool one) and it seems you can do what you need there as well.
Copy the contents of the text file and then in the hexed.it site:
- create a new empty file
- press ctrl-v to paste 
- on the popup, you select `Insert the data at the cursor position` and in the `Please specify how the data should be interpreted` you choose the `Hexadecimal Values` option
- after that, you press the "Export" button and you save it as a .bin file

dealaddict, 2019-02-11:
I got it working. Thanks for being so patient with me. I owe you a lot for learning these tools. Cheers man! @bogiton

closed 2019-02-11 by dealaddict

iceman1001, 2019-02-13:
@bogiton   create dumps from text..  what different from the eml format was his text dumps?

bogiton, 2019-02-13:
@iceman1001 Hmm, I suppose no different at all :P
Didn't recall we had that python converter script in the Tools.
Well, at least I learned about hexed.it!

iceman1001, 2019-02-13:
yeah,  I took them from proxmark project since they was made in python in order to convert easier.  Before the GUI was done.    However create / edit a single file...    but thats what usually is done with a hexeditor do we need to built in that functionality?  
So to recapture this issue.    

1. had a text dump from an app.  (is actually in EML format)?
2. could have used the latest GUI for converting it to BIN
3. upload it to chameleon
4. download it from chameleon  (eml/bin formats)
5. did @dealaddict  want to use with the app again??

bogiton, 2019-02-13:
So, as I just checked, you can get it done simply if you rename those txt dumps to .eml and then convert to bin using the Chameleon GUI (dump management tab). It's been a while since I tested the GUI's new features :)

iceman1001, 2019-02-13:
So we have full functionality.  Good,  time to put to rest.

iceman1001, 2019-02-26:
I guess this issues also was struck  by latest fix on uploading MFU / ntag (bin/eml) dumpfiles without header field.  Might just test again.