iceman1001/ChameleonMini-rebootedGUI #100: Chameleon Rev G dump problems

ph553, 2019-02-26:
Hi
I'm currently trying to use the GUI for my rev G, but i've encountered some problems

First of all, when I try to upload a dump to a slot, i get something like the slot 1 in the following image:
[image]
while it uploads correctly using teraterm(slot 2). If i read the tag on slot 1, i get an empty card with uid 0000000

Second problem would be while downloading a dump.
i get something like
[image]
Where the real card would start at the 48th byte
the data from the 112th bytes would be the counters(i guess)

The card data shows empty when read since it is a MIFARE ULTRALIGHT and therefore is only supposed to have 64 bytes of data

Maybe the GUI is writing/checking data at the wrong address?

bogiton, 2019-02-26:
Hello @ph553 
Can you please share the first dump you are trying to upload? Are you sure that it doesn't begin with the leading zeroes?

ph553, 2019-02-26:
[card.txt](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/2906513/card.txt)
It was originally in .bin format but i couldnt upload it as a bin file

bogiton, 2019-02-26:
Ok, it seems that neither in RevE works as it should.
Until, it is fixed, the quick and dirty way is to add some zeroes in the beginning of the dump to upload.
I tried that to your card.bin (find it attached) and worked for me.
[card.zip](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/2906619/card.zip)

iceman1001, 2019-02-26:
aha, I see,  its the binary loader / eml loader that doesn't different between a proxmark3 dump with header or not...   It assumes its a dump file without headers..

        private static byte[] ReadFileIntoByteArray(string filename)
        {
            var fi = new FileInfo(filename);
            if (fi.Exists)
            {
                var dumpStrategy = DumpStrategyFactory.Create(fi.FullName);
                var data = dumpStrategy.Read();
                if (data.Length < 1024 && data.Length != 320)
                    data = data.Skip(MifareUltralightCardInfo.PrefixLength).ToArray();
                return data;
            }
            return null;
        }

---
        public override byte[] Read() =>
            File.ReadAllBytes(FileName);

We would need to add some kind of logic there.

iceman1001, 2019-02-26:
The logic of  less than 1024 and not 320,  then skip header,  isn't good enough I guess.
We would need to match more lengths...

iceman1001, 2019-02-26:
This patch tries to solve BIN/EML based uploads with dump files that has or has not a header field inside.
ie old dumpfiles from other devices or older proxmark clients.  

https://github.com/iceman1001/ChameleonMini-rebootedGUI/commit/cc761f11ae44b592f9c09a6c972d526cbf1054b7

@ph553   How about you test it out?

ph553, 2019-02-26:
@iceman1001 just tested the latest version on both my revG and revE-rebooted.
Upload works correctly, but if i download a dump, it still put some bytes filled with 0. I tried reuploading from a dump, but each time i upload-download, it offsets the card data more.

iceman1001, 2019-02-27:
Aha,  I didn't look at the download,   I am off so I will take a look at when I return..  or @bogiton  may do some magic

iceman1001, 2019-03-07:
i think i will add a check if header is all zeros..

iceman1001, 2019-03-11:
test again @ph553    v1.2.0.18  should solve it.

ph553, 2019-03-11:
@iceman1001
It works correctly for me, thanks!

closed 2019-03-11 by iceman1001