iceman1001/ChameleonMini-rebootedGUI #103: Drop and drag file onto slot and upload

iceman1001, 2019-03-12:
**Is your feature request related to a problem? Please describe.**
I want to be able to drop and drag a bin/eml/json file onto the operation tab and one slot.

**Describe the solution you'd like**
When I drop and drag a bin/eml/json file,  and I want to see a hover effet which slot I am going to upload file to.  When releasing mouse,  it should upload and autodetect type/settings for the slot.  Like dropping a mifare classic 1024byte file, should upload and set Mode to **MF_CLASSIC_1K** etc. 

**Additional context**
I think this will give a smooth nice user experience how to operate their Chameleon mini device.

bogiton, 2019-03-13:
I have the drag-drop upload functionality for bin files ready.
I believe that the eml/json support for upload, as well as the autodetection of the file type should be added in the UploadDump method internals.
What do you think?

iceman1001, 2019-03-13:
yup yup

iceman1001, 2019-03-13:
The drag-and-drop check should also take in consideration that ntag/ul may or may not have a 48byte header...

iceman1001, 2019-09-14:
@McEloff  did you solve this?