iceman1001/ChameleonMini-rebootedGUI #110: easy choosing of restore fw

iceman1001, 2019-09-23:
Todays function with restoring the fabric firmware is included.  It is however a bit limited given we have a more updated firmware available.

I suggest we add a way to search the Rebooted-repo releases for the latest version (which is not auto created by the CI )  and give the user a easy way to select which firmware they would like to have flashed.  Maybe a dropdown?

[image]

bogiton, 2019-09-29:
I was looking at that yesterday, but there are a couple of drawbacks. For example, the only release for the time being, after excluding the untagged, is the "v1.0" one. But the first release doesn't contain the hex/eep files, to automatically download and flash via the GUI. We either wait for the new release or we also include the untagged ones in the list.

iceman1001, 2019-09-29:
Yup.  Not only that.  We need to be able to convert the two files to the format the bootload.exe wants.

Post-processes for travis to do it automatically?

bogiton, 2019-09-29:
We could just issue the Createbin.exe and avr-objcopy.exe from within the GUI to generate the bin files for BOOT_LOADER_EXE and leave the hex/eep files as they are.

iceman1001, 2019-09-30:
yup,  then we would need to distribute some more files.

iceman1001, 2019-12-20:
@grspy  I saw you added code but I didn't see the GUI changed.  We would need clear indications to what will happen to ensure correctness.

grspy, 2019-12-20:
I didn't add an option to choose a firmware to flash, just the automatic recognition of the device in order to flash the stock firmware using the "Load Defaults" button. Now this button can also be used to flash a RevG device with the included stock firmware.