iceman1001/ChameleonMini-rebootedGUI #111: draft of v1.3

iceman1001, 2019-09-23:
A draft for v1.3 release of the GUI

@shinhub @bogiton @slurdge

Feel free to update the text to cover all changes.

iceman1001, 2019-09-30:
After testing a bit,  I must say the user experience left me with a bad taste in my mouth.
The gui was slow. Really slow.   The selected slot jumped between operations,  the detection code didn't work,  the device went offline / online.   It was a bizarr experience.   Now if this depends on device fw or gui or a combi I dont know.  Regardless this needs more proper testing.

shinhub, 2019-11-03:
@iceman1001 I think I fixed most of slowness (#112) as well as a slot index selection issue. But this one should still be slower than oldest GUIs because:
1) it checks UID and UIDSIZE based on real FW content to ensure we set a consistent one when switching from confs, most notably from 4 to 7bytes UID (before that, the GUI sometimes appeared with an UID that was not the one in FW memory)
2) clear now really clears whole slot memory (where before, it depended on which bug you triggered), and sometimes slots memories are bigger than before on FW (so, more memory to clear)

As for DETECTION, I think it it should be from FW side, I will look into it. But anyway, DETECTION has always been a pain lately.

shinhub, 2019-11-03:
Plus there were some bugs in original code on Serial data reading, that were very visibly triggered (introducing delay till a 1s manual timeout is reached) by an improper use of "SendCommand" I introduced when checking UID sizes. See #113. This should speed everything back to when checks were not made on UID, slots were not reset consistently with FW, and all memory was not used/cleared.

iceman1001, 2019-11-04:
@shinhub   Don't forget to add lines in Changelog.md...

iceman1001, 2019-12-20:
@shinhub   @grspy  @bogiton   
With @grspy 's commits we are getting to a much smoother experience.
How about you try it out and give some feedback on it?

iceman1001, 2020-01-04:
Thanks to @merlokk  we now have Appveyor CI builds.   I hooked up the badge on readme,  looking good.

closed 2020-05-20 by iceman1001