iceman1001/ChameleonMini-rebootedGUI #117: GUI stops responding after mfu dump upload

BreakSecurity, 2019-11-12:
**Describe the bug**
GUI stops responding after mfu dump upload in any slots. Needs to force close and reopen.

**To Reproduce**
Steps to reproduce the behavior:
1. Select any slot
2. Click on 'Upload Dump'
3. Select the attached dump (json listed below)
4. See GUI freeze

Json:
```
{
  "Created": "proxmark3",
  "FileType": "mfu",
  "Card": {
    "UID": "04481AD2A05E85",
    "Version": "0004030101000B03",
    "TBO_0": "0000",
    "TBO_1": "00",
    "Signature": "B44E1D9A80C825E1F7AE32506848DC6C10CE7F5DFA835158D52B1A0B7C4ED470",
    "Counter0": "000000",
    "Tearing0": "BD",
    "Counter1": "000000",
    "Tearing1": "BD",
    "Counter2": "000000",
    "Tearing2": "BD"
  },
  "blocks": {
    "0": "04481ADE",
    "1": "D2A05E85",
    "2": "A948F203",
    "3": "7FFFFFFE",
    "4": "01030000",
    "5": "0101012E",
    "6": "76B7A400",
    "7": "00AE10A8",
    "8": "13015C46",
    "9": "84AABA40",
    "10": "00000000",
    "11": "00000000",
    "12": "00000000",
    "13": "00000000",
    "14": "00000000",
    "15": "00020000",
    "16": "000000FF",
    "17": "00050000",
    "18": "00000000",
    "19": "00000000"
  }
}
```

iceman1001, 2019-11-12:
@shinhub something to weed out?

shinhub, 2019-11-12:
@iceman1001 The method call `WaitForResponseFromReceiver.WaitOne();`in `XMODEM.cs`(line 1271) hangs indefinitely with this ULTRALIGHT_EV1_80B JSON dump.
The dump get properly updated though, but GUI is perpetually stuck.
Tried to upload other dumps for Mifare, including JSON, and did not have the same issue.
The issue triggers with ULTRALIGHT_EV1_80B dump upload as far as I tested, with this JSON, or with a binary format of the same content.
It seems that `var bytes = dump.Data.Concat(dump.Extra).ToArray();`in `FrmMain.cs`(line 2089) is returning way too much data for a 80Bytes dump though (100+ bytes), so the bug might be there, or in `var dump = ReadFileIntoByteArray(filename);`(line 2080) just above, that might just not support 80Bytes Ultralight at all...

iceman1001, 2019-12-11:
@shinhub   Did you find some solution for this?

grspy, 2019-12-13:
I just tested what @shinhub suggested and it's in fact the appended "Extras" (52 additional bytes) that blocks the upload for the ULTRALIGHT_EV1_80B configuration. This is not happening when the config is ULTRALIGHT_EV1_164B, so I suppose that there is something wrong with mem size of the 80B ULTRALIGHT_EV1 in the firmware. If you omit the Extras, the upload on all slots works fine.

iceman1001, 2019-12-20:
And its only for RevE rebooted?  not for RevG?

grspy, 2019-12-20:
As far as I have tested, yes. That's why I believe the issue is in the RevE rebooted firmware.