iceman1001/ChameleonMini-rebootedGUI #118: GUI overwrite downloaded dumps if more slots share the same UID

BreakSecurity, 2019-11-12:
**Describe the bug**
GUI overwrite downloaded dumps if more slots share the same UID. E.g. if I upload the UID xxx to slot 1, yyy to slot 2, and xxx to slot 3 when I download all the dumps I will obtain only two files instead of three files (xxx is overwritten, even if different manipulation were done by the reader to the two slots)

**To Reproduce**
Steps to reproduce the behavior:
1. Upload the same dump to two different slots
2. Perform different action on the slots
3. Click on 'Select All'
4. Click on 'Download Dump'
5. Select a folder
6. See error (we lost one of the two dump)

**Expected behavior**
A suffix should be added to the UID. E.g. xxx_slot_n (where UID = xxx and n is the slot number)

**Desktop (please complete the following information):**
 - OS: Windows 10

iceman1001, 2019-11-12:
Hm,  never thought of having the same dump on multiple slots.  I agree it has a definite value to work like that.  The file saving should atleast a _(n)  suffix to the filename where n is last existing number plus one.

The idea with slot number also works but gets a bit confusing after awhile.  I rather not see GUI usage reflect on filenames.  Ie,  use on slot 1 or slot N,

iceman1001, 2019-12-11:
@BreakSecurity   You can test the fix.   I will close this one.

closed 2019-12-11 by iceman1001

BreakSecurity, 2019-12-13:
Do not work...

Selected slot from 4 to 8 and tried to upload an MFU dump GUI freeze as in #117 and only the last slot was effectively written

iceman1001, 2019-12-13:
this issue is about **downloading** dumps.

BreakSecurity, 2019-12-13:
Sorry I commented the wrong issue.

Regarding this one it works pretty good, but to be perfect it should ask the user the desidered output folder and the format (at the moment they are saved in batch to the program folder in .bin format)

iceman1001, 2019-12-13:
Sounds fair and intersting features.  Feel free to contribute!