iceman1001/ChameleonMini-rebootedGUI #119: [REQ] Option to uploads to all slots the same dump

BreakSecurity, 2019-11-12:
Would be nice to have the possibility to uploads to all slots the same dump. It could be useful in field testing when I have to guess what the reader is doing to the tag. 

E.g. I have some checksum computed on the base of some tag information and timestamp, to reverse the operations done by the reader could be useful to use the same dump in different times.

closed 2019-12-09 by iceman1001

BreakSecurity, 2019-12-13:
Selected slot from 4 to 8 and tried to upload an MFU dump GUI freeze as in #117 and only the last slot was effectively written

grspy, 2019-12-13:
Every time I tried with a .bin dump, it worked fine. Are you still trying to mass upload that json file? Can you please convert it first to .bin using the file manager (Utils tab) and try with that?

BreakSecurity, 2019-12-13:
Same problem with the .bin file attached

[test.zip](https://github.com/iceman1001/ChameleonMini-rebootedGUI/files/3961136/test.zip)

commit ca9a60f1 (2019-12-09): Merge pull request #123 from grspy/master

Upload a dump to multiple slots (fix #119)