iceman1001/ChameleonMini-rebootedGUI #137: How to GETUID

Bowski1981, 2020-03-27:
Hi All,

I am using the Chameleon Tiny for a few days and I am not used to it yet.
I managed to use some basic functions and get the connection to my Mac working properly.

There is something I am just wondering about... When reading an UID by GETUID and copying this to another slot, the UID changes. So the UID which is shown in my Tiny is different from the one I am reading from the card directly. Even when using the clone command it differs from the original UID every time.

Am I missing something? Or is it just normal?

iceman1001, 2020-03-28:
Start with post all different steps you did,  otherwise noone can follow your issue.

And don't randomly label issues. We, the maintainers,  do the labeling.

Bowski81, 2020-04-15:
Thanks a lot for your answer and sorry for any inconveniences caused.

I followed the first step guide on GitHub to setup my chameleon tiny.

It is working properly and I am able to read several tags and get the correct UID.

However, while trying to simulate an ISO14443 Tag and setting the UID it won't let me change the UID.

Here is a screenshot of the problem:
[image]

As you can see I'm trying to change the UID. But it does change the UID every time. I am not able to change it to the value I want.

Am I missing something?

iceman1001, 2020-04-16:
Chameleon Tiny is a RevG based device.  This is the GUI project,   in what why does the GUI not work for you?  Otherwise you should direct your question to the Rfidresearchgroups' firmware repo for RevG / Tiny,   ref:   https://github.com/RfidResearchGroup/ChameleonMini

closed 2022-07-14 by iceman1001