iceman1001/ChameleonMini-rebootedGUI #139: HELP text truncated

iceman1001, 2020-05-20:
**Describe the bug**
The output from the HELP command inside tab serial,  should print all available commands.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to TAB SERIAL
2. Click on HELP
3. Client on SEND
4. See error at the end of the output from that command.

**Expected behavior**
To list the same amount of detected available commands as to list to the left on tab serial.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

Loupetre, 2020-11-25:
Hello,

At the line 1798 of the FrmMain.cs file, just change 
int blockLimit = 275;
by
int blockLimit = 375;

The typical answer's lenght of the HELP commande is 347 bytes.

Here is the result after changing this value :

[image]

iceman1001, 2020-11-25:
Great catch!
Make a PR?

closed 2020-11-26 by iceman1001

Loupetre, 2020-11-26:
Sorry, I see you've made the change before I succeeded in understanding what is a PR and how to do it : I've never made anything like that :)
I gonna go for a try with another small enhancement (french language, that is my cup of tea).

commit 57904435 (2020-11-26): Update FrmMain.cs

fix #139
increase download block size in order to get all text response from device when issuing "HELP",    Thanks @Loupetre
commit a1ca5277 (2020-11-26): Update FrmMain.cs

fix #139
increase download block size in order to get all text response from device when issuing "HELP",    Thanks @Loupetre