iceman1001/ChameleonMini-rebootedGUI #140: Giant text on high-resolution display makes Slot Configuration unreadable

bmorgenthaler, 2020-06-30:
The text size does not appear to adjust with display settings and defaults to a giant size making all the slot configuration information unreadable.

Steps to reproduce the behavior:
1. Open application on high resolution (3440x1440 or higher) desktop

**Expected behavior**
Text resizes with OS settings for Ease of Access, Font Scaling, etc. and is not cut off.

**Screenshots**
[image]

**Desktop**
 - OS: Windows 10 Parallels VM on MacOS 10.15.5

bmorgenthaler, 2020-06-30:
Nevermind, magically fixed itself.

closed 2020-06-30 by bmorgenthaler

iceman1001, 2020-06-30:
The gui doesn't enforce any font sizes.  It should adapt to Windows settings.