iceman1001/ChameleonMini-rebootedGUI #145: Setting slot 8 as active reverts after refresh

sawft99, 2022-06-03:
**Describe the bug**
In the GUI and using the built in terminal, if you set slot 8 as active it will take but refreshing or applying the config will revert it to the last slot you had active before that

**To Reproduce**
Steps to reproduce the behavior:
1. Select any slot other than 8 and set it as active with the built in terminal or GUI
2. If you like, refresh/apply/reboot/reconnect/etc...
3. Set slot 8 to active
4. The GUI will show the slot as changing and even the lights on the Chameleon change
5. Refresh/apply/reboot/reconnect/etc...
6. The active slot is reverted back to the last slot used before trying to make #8 active
7. The lights also revert on the Chameleon

**Expected behavior**
   
   - For slot 8 to stay as active after refreshing/applying/etc

**Desktop:**
 
   - OS: Windows 11 x64
   - App: 1.3

**Device & Firmware**

 - Chameleon Tiny Pro
 - [RfsidResearchGroup Build 17f32094d905d6ef08fb4e73cbc62813a2a21b74]https://github.com/RfidResearchGroup/ChameleonMini/releases/tag/Build-17f32094d905d6ef08fb4e73cbc62813a2a21b74

iceman1001, 2022-07-14:
There is no active code contributors to this repo any longer.   So feel free to contribute,  be the change you want to be

Loupetre, 2022-07-18:
Hello,

I think I have found the correction to be done. I cannot do a pull request because I have proxy limitation @work.

In **FrmMain.cs**, function **RestoreActiveSlot()**, juste change : 

if (_active_selected_slot > _tagslotIndexOffset+7) _active_selected_slot = **_tagslotIndexOffset+7**;
by
if (_active_selected_slot > _tagslotIndexOffset+7) _active_selected_slot = **8 -_tagslotIndexOffset**;

closed 2022-07-18 by iceman1001

iceman1001, 2022-08-06:
@Loupetre   I added your suggestion,   if you want to verify it

commit 6be26794 (2022-07-18): suggested fix #145
Active slot8 reverts.

Plus upgraded dotnet to 4.7.2 and new version number.